greatscottgadgets/packetry #19: Something going wrong with the mapping between widgets and items

martinling, 2022-04-12:
I think something is going wrong involving the view's reuse of the widget pool. There are two symptoms:

1. Sometimes, clicking to expand an item will also expand one or more other items nearby.
2. I just managed to trigger the following panic - this was on commit [`10f95f5`](https://github.com/martinling/packetry/commit/10f95f5e950201fb64ae346f8c9291fde51427df) (which includes some WIP but not in the UI code):

```
$ RUST_BACKTRACE=1 cargo run --release ../luna-analyzer-capture/hackrf-mac-sweep.pcap
   Compiling packetry v0.1.0 (/home/martin/repos/packetry)
warning: field is never read: `descriptor_index`
   --> src/capture.rs:426:5
    |
426 |     descriptor_index: HybridIndex,
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: `#[warn(dead_code)]` on by default

warning: `packetry` (bin "packetry") generated 1 warning
    Finished release [optimized] target(s) in 4.67s
     Running `target/release/packetry ../luna-analyzer-capture/hackrf-mac-sweep.pcap`
Storage summary:
  Packet data: 270.66 MiB
  Packet index: 3,557,371 values in 4,321 entries, 6.91 MiB
  Transaction index: 1,554,963 values in 13,305 entries, 1.90 MiB
  Transfer index: 2,404 entries, 18.78 KiB
  Endpoint states: 11,975 entries, 11.69 KiB
  Endpoint state index: 2,404 values in 47 entries, 3.77 KiB
  Endpoint transaction indices: 1,554,963 values in 12,395 entries, 1.85 MiB
  Endpoint transfer indices: 1,203 values in 36 entries, 2.26 KiB
Total overhead: 4.4% (11.79 MiB)
thread 'main' panicked at 'The item has to exist.', src/main.rs:90:18
stack backtrace:
   0: rust_begin_unwind
             at /usr/src/rustc-1.57.0/library/std/src/panicking.rs:517:5
   1: core::panicking::panic_fmt
             at /usr/src/rustc-1.57.0/library/core/src/panicking.rs:100:14
   2: core::panicking::panic_display
             at /usr/src/rustc-1.57.0/library/core/src/panicking.rs:64:5
   3: core::option::expect_failed
             at /usr/src/rustc-1.57.0/library/core/src/option.rs:1638:5
   4: gtk4::auto::signal_list_item_factory::SignalListItemFactory::connect_bind::bind_trampoline
   5: g_cclosure_marshal_VOID__OBJECTv
   6: <unknown>
   7: g_signal_emit_valist
   8: g_signal_emit
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: <unknown>
  13: g_closure_invoke
  14: <unknown>
  15: g_signal_emit_valist
  16: g_signal_emit
  17: <unknown>
  18: g_closure_invoke
  19: <unknown>
  20: g_signal_emit_valist
  21: g_signal_emit
  22: gtk_tree_list_row_set_expanded
  23: gtk4::auto::expander::Expander::connect_expanded_notify::notify_expanded_trampoline
  24: g_closure_invoke
  25: <unknown>
  26: g_signal_emit_valist
  27: g_signal_emit
  28: <unknown>
  29: g_object_notify
  30: <unknown>
  31: g_signal_emit_valist
  32: g_signal_emit
  33: gtk_widget_activate
  34: <unknown>
  35: <unknown>
  36: g_signal_emit_valist
  37: g_signal_emit
  38: <unknown>
  39: g_cclosure_marshal_VOID__BOXEDv
  40: <unknown>
  41: g_signal_emit_valist
  42: g_signal_emit
  43: <unknown>
  44: <unknown>
  45: <unknown>
  46: <unknown>
  47: <unknown>
  48: <unknown>
  49: <unknown>
  50: <unknown>
  51: <unknown>
  52: <unknown>
  53: g_closure_invoke
  54: <unknown>
  55: g_signal_emit_valist
  56: g_signal_emit
  57: <unknown>
  58: <unknown>
  59: g_main_context_dispatch
  60: <unknown>
  61: g_main_context_iteration
  62: g_application_run
  63: <O as gio::application::ApplicationExtManual>::run_with_args
  64: packetry::main
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.

miek, 2022-04-12:
I wonder whether it might've been introduced in #17? Maybe it needs to explicitly disconnect previous handlers?

martinling, 2022-04-13:
Yes, I think that's the issue. We need to store the `SignalHandlerId` returned by the connect call, and then either add an `unbind` handler to the factory which disconnects it, or disconnect it in the `bind` handler before reuse it.

PR #25 (2022-05-09): Fix reuse of Expander widgets, without unsafe blocks or overhead
This stacks on #24, replaces #21 and closes #19.

The basic idea is to use a subclass instead of an `Expander` directly, so that we can attach arbitrary data to it.

The gtk-rs bindings don't allow subclassing `Expander`, but we can make a subclass of `gtk::Box` that just has one `Expander` inside it, and use that as our widget. Then we can keep the `SignalHandlerId` in the internal struct for the subclass.

This does add one extra layer of widget nesting, but since we were putting the `Expander` inside a box with the `Label` wid [... cut]
closed 2022-05-09 by miek