greatscottgadgets/packetry #42: packtry roadmap for windows 10 ?

d-c-d, 2022-10-27:
I see [GUI development with Rust and GTK 4 (Windows)](https://gtk-rs.org/gtk4-rs/stable/latest/book/installation_windows.html) but the readme page only mentions linux.  Is there a roadmap, hope or plan with respect to getting packetry to run on Windows 10?

Are there any design constraints that prevent this from happening?

martinling, 2022-11-01:
I've been building and testing on Windows 10, everything works fine. I'll see about adding some build instructions to the readme.

martinling, 2022-11-12:
As a quick guide to building on Windows right now:

First, you need to get to the point where you can build `gtk4-sys` and the other related gtk-rs sys packages, which requires you to have a working `pkg-config` that can find all the GTK dependencies.

The best way to do that at the moment is to follow the installation instructions on the page you mentioned at [GUI development with Rust and GTK4 for Windows](https://gtk-rs.org/gtk4-rs/stable/latest/book/installation_windows.html).You should be able to use either the MSVC or GNU toolchain routes, according to your preference.

Then, check out my branch for PR #45, which eliminates the libpcap/wpcap dependency. Once that PR is merged this will no longer be necessary.

You should then be able to build Packetry as normal with `cargo build`.

This combination is working for me, but please report here if you have problems.

PR #46 (2022-11-14): Update README with screenshot and further instructions.
Closes #42 and closes #44.