greatscottgadgets/packetry #68: Tool Bar Icons

NetSpida, 2023-03-22:
Currently, Packetry has a toolbar with the following Icons (in that order):

"Load PCAP file", "Save PCAP file", "Start Capture", "Stop Capture", "Select Capture Speed"
[image]

I would like to put a few points up to discussion:

1. Maybe a dropdown box to select the capture device could be added. This would give visual feedback whether or not a cynthion device was recognized and possibly to choose between several capture devices similar to the [sigrok pulseview UI](https://sigrok.org/wiki/File:PulseView_I2C_DS1307_Decode.png).
2. "Load PCAP file" might be an entry in that device selection dropdown, too, although I guess it might be hard to find in that list.
3. The "Start Capture" and "Stop Capture" buttons could be merged into one, similar to the [sigrok pulseview UI](https://sigrok.org/wiki/File:PulseView_I2C_DS1307_Decode.png)
4. Possibly reorder the toolbar icons in the order they are usually used:
"Load PCAP file"/"Select Capture device", "Select Capture Speed", "Start Capture", ("Stop Capture"), "Save PCAP file"

These suggestions will likely need discussion and are probably subject to taste.

miek, 2023-03-22:
> I think we should keep opening files and capturing from devices separate. They're quite different operations conceptually, even if they function quite similarly.

I agree with that and in general I think it's more common to have icons grouped together by category, so I'd prefer to do that. PulseView is doing the same: grouping together file operations, zoom operations, and live capture operations.

> I am opposed to start/stop capture being the same button, because there are cases where the capture will stop without user interaction (e.g. buffer full), in which case the button may change from "stop" to "start" just before the user clicks it, causing a new capture to be started unintentionally. This has caused me to lose captures in PulseView multiple times, and in fact I've had a [request for separate buttons](https://sigrok.org/bugzilla/show_bug.cgi?id=634) in PulseView open since 2015.

:+1: 

> We also have the question of what to put on each side of the application title in the header bar, and the things on each side need to balance somewhat. Or we could move away from the GTK convention of a centred title there.

This raises another point that I wanted to talk about: right now we're using a HeaderBar to merge icons in with the title bar, but I find this quite awkward as I often tile a few windows side-by-side, then full-screen one as needed - this means that the title bar goes away and takes away all the controls with it. I'd personally prefer to switch to a normal toolbar.

> **Packetry** | open_file_folder floppy_disk | [Device arrow_down] [Speed arrow_down] record_button stop_button

:+1:

antoinevg, 2023-03-22:
> **Packetry** | 📂 💾 | [Device ⬇️] [Speed ⬇️] ⏺️ ⏹️

If the device and speed dropdowns will always have a constant width this works very well.

However, if they could get resized according to content at any point I would much prefer to see the Start/Stop buttons appear to their left for the sake of muscle memory!

martinling, 2023-04-29:
Implemented in #73. I think we can close this now.

closed 2024-01-09 by martinling