greatscottgadgets/packetry #74: GTK 4.10.1 Dropdowns are buggy on macOS

antoinevg, 2023-04-17:
### Environment

* Packetry version: [4f49e7f](https://github.com/greatscottgadgets/packetry/commit/4f49e7fc7e168bf7802168ffa329c076f97ff3b0)
* GTK version: `4.10.1`
* macOS version: `13.3 (Ventura)`

### Expected Behaviour

Clicking on an entry in the toolbar's speed dropdown should select the entry and trigger a GTK event.

### Actual Behaviour

Clicking on an entry closes the dropdown, does not change the selection and does not trigger a GTK event.

### Workaround

* Select the dropdown
* Navigate to the entry via keyboard cursor keys
* Hit Enter

antoinevg, 2024-01-04:
I'm not sure when exactly it was fixed but is no longer an issue on newer gtk releases.

closed 2024-01-04 by antoinevg

mossmann, 2024-07-15:
@antoinevg Did you happen to upgrade your OS between the time when you had this issue and when it disappeared?

antoinevg, 2024-07-16:
I tend to keep my work machine on the latest release so that is very likely!