greatscottgadgets/packetry #121: Packetry does not follow system-level dark mode on Windows and macOS

martinling, 2024-07-11:
On Windows and macOS, Packetry appears in a light theme even if dark mode is selected globally in the OS.

It's not clear to me yet how this is supposed to work - I'd have thought it was GTK's responsibilty and shouldn't require code in the application.

Perhaps we'd need to use [`libadwaita::Application`](https://world.pages.gitlab.gnome.org/Rust/libadwaita-rs/stable/latest/docs/libadwaita/struct.Application.html) instead of `gtk::Application`? But that seems wrong, as we're trying to produce a generic cross-platform application, not a GNOME one.

probonopd, 2024-08-07:
Please don't. [libadwaita doesn't work well](https://www.youtube.com/watch?v=WFqJA1q-IKY) outside of the Gnome desktop.  Switching to libadwaita would significantly reduce the potential user base for this application, essentially limiting it to the fraction of users who happen to use Gnome.

martinling, 2024-08-07:
Yeah, it's not a route we want to take. But it seems that's where they've put the code that handles this.

So we're going to need to either re-implement that functionality ourselves, or find something else we can pull in to take care of it.

martinling, 2024-08-08:
It looks like we might be able to do this with a combination of the [rust-dark-light](https://github.com/frewsxcv/rust-dark-light) crate to detect the system setting, and [gtk::Settings::set_gtk_application_prefer_dark_theme](https://gtk-rs.org/gtk4-rs/stable/latest/docs/gtk4/struct.Settings.html#method.set_gtk_application_prefer_dark_theme) to configure the GTK behaviour.

PR #166 (2024-08-09): Select dark theme if detected as a system preference.
Uses:
- Rust [dark-light](https://github.com/frewsxcv/rust-dark-light) crate to detect the system setting
- GtkSettings [set_gtk_application_prefer_dark_theme](https://gtk-rs.org/gtk4-rs/stable/latest/docs/gtk4/struct.Settings.html#method.set_gtk_application_prefer_dark_theme) to configure the GTK behaviour.  

Closes #121.
closed 2024-08-09 by miek