greatscottgadgets/packetry #122: Analyzer captures own analysis traffic

mossmann, 2024-07-11:
If Cynthion's CONTROL port is connected to the same hub as TARGET, CONTROL traffic is observed on TARGET, resulting an infinite loop. This appears as many "invalid groups" in Packetry:

[image]

Note that adjacent USB ports on some computers are connected to an internal hub, so this can happen even when not using an external hub.

Workaround: Connect CONTROL to a USB port with a different host controller than the host port connected to TARGET. Most computers have more than one USB host controller. Typically ports that are physically distant from one another have separate host controllers. One way to be absolutely certain is to use two separate computers, one for CONTROL and one for TARGET.

Various solutions are possible, probably involving implementation of a capture filter in the analyzer gateware and some way to control it from within Packetry.

mossmann, 2024-07-18:
Even if we implement capture filtering to solve this problem, there would still be failures due to the bandwidth limitation when a high-throughput target device is being captured. We will always want to advise users to avoid this setup whenever possible, and we may want to try to detect the scenario and warn the user.

martinling, 2024-08-09:
I think this issue should be moved to the Cynthion repo. Any solution will need features in the analyzer first before we can support them in Packetry.