greatscottgadgets/packetry #129: Drop-down menus broken on macOS

mossmann, 2024-07-14:
The capture speed is stuck at the default High Speed (480 Mbps). The drop-down list appears when clicked, but selection of a different speed fails.

console output:
```
UNSUPPORTED (log once): POSSIBLE ISSUE: unit 1 GLD_TEXTURE_INDEX_2D is unloadable and bound to sampler type (Float) - using zero texture because texture unloadable

(<unknown>:11646): Gdk-CRITICAL **: 12:55:47.960: _gdk_macos_surface_update_fullscreen_state: assertion 'GDK_IS_MACOS_SURFACE (self)' failed

(<unknown>:11646): Gtk-WARNING **: 12:55:47.968: Broken accounting of active state for widget 0x154611c80(GtkBuiltinIcon)

(<unknown>:11646): Gtk-WARNING **: 12:55:47.968: Broken accounting of active state for widget 0x154611590(GtkBox)

(<unknown>:11646): Gtk-WARNING **: 12:55:47.968: Broken accounting of active state for widget 0x154611410(GtkToggleButton)

(<unknown>:11646): Gtk-WARNING **: 12:55:47.968: Broken accounting of active state for widget 0x1546111c0(GtkDropDown)

(<unknown>:11646): Gtk-WARNING **: 12:55:47.968: Broken accounting of active state for widget 0x154614200(GtkBox)

(<unknown>:11646): Gtk-WARNING **: 12:55:47.968: Broken accounting of active state for widget 0x144616900(GtkBox)

(<unknown>:11646): Gtk-WARNING **: 12:55:47.968: Broken accounting of active state for widget 0x144616f00(GtkCenterBox)

(<unknown>:11646): Gtk-WARNING **: 12:55:47.968: Broken accounting of active state for widget 0x144616510(GtkRevealer)

(<unknown>:11646): Gtk-WARNING **: 12:55:47.968: Broken accounting of active state for widget 0x144616070(GtkActionBar)

(<unknown>:11646): Gtk-WARNING **: 12:55:47.968: Broken accounting of active state for widget 0x144713ea0(GtkBox)

(<unknown>:11646): Gtk-WARNING **: 12:55:47.968: Broken accounting of active state for widget 0x1446142c0(GtkApplicationWindow
```
The initial `GLD_TEXTURE_INDEX_2D` message happens when the Packetry window first appears. The remaining messages happen each time the Speed Drop-down is clicked.

Similarly the About window does not appear when selected from the drop-down menu, and selection of a non-default capture device fails if there are multiple capture devices listed in the device selector.

`packetry --version` output:
```
Packetry version 0.1.0 (git fc7d276)

Runtime information:
  GTK version: 4.14.4

Packetry build information:
  Git commit: fc7d276efae23c5c89e8ca92c88766f61206f6ed
  Cargo package version: 0.1.0
  Enabled features: (none)

Rust compiler:
  Version: rustc 1.79.0 (129f3b996 2024-06-10)
  Target: aarch64-apple-darwin (little-endian, 64-bit)
  Optimization level: 3
  Debug build: no
```

martinling, 2024-07-14:
This really seems like a GTK bug, not a Packetry one. I'm sure I've had that dropdown working previously on macOS.

We may be able to bisect GTK to figure out where it broke.

mossmann, 2024-07-15:
> This really seems like a GTK bug, not a Packetry one. I'm sure I've had that dropdown working previously on macOS.

Have any idea what GTK version you had success with?

martinling, 2024-07-15:
GTK 4.8.1. I've just built current `main` against it and the dropdown works.

mossmann, 2024-07-15:
Looks like this bug was seen before, but we closed it because it seemed to have been fixed: https://github.com/greatscottgadgets/packetry/issues/74

martinling, 2024-07-15:
It looks like perhaps #74 was encountering a _previous_ occurence of the same bug, which was reported upstream in 4.10 as https://gitlab.gnome.org/GNOME/gtk/-/issues/5905 and was fixed as of 4.12.

mossmann, 2024-07-15:
I reproduced the problem with GTK 4.12.5.

mossmann, 2024-07-15:
Thanks to those upstream bug reports, I have successfully tested a workaround (on 4.12.1 and 4.14.4): For the speed drop-down (or any drop-down containing more than one item) you can use the keyboard instead of the mouse.

depili, 2024-07-15:
On M1pro MBP with Macos 14.5 Sonoma (The latest release as of this message) I get the accounting errors in the console, but otherwise the dropdowns work and change the speed, as per the GTK issue linked above.

mossmann, 2024-07-15:
@depili Thanks for the report! Based on this and the upstream reports, it seems that this issue may go away with an OS update.

villu164, 2024-07-15:
I can confirm, that now I can select the 12Mbps option with keyboard workaround and I can capture Wireless mouse packets

https://github.com/user-attachments/assets/c15859d6-ff41-4db5-8ffe-c2e7b9139d6f

mossmann, 2024-07-16:
I upgraded my OS to 13.6.7 with no change. Then I upgraded it to 14.5, and it fixed the problem.

mossmann, 2024-07-16:
The upstream GTK bug(s) appear to affect only macOS 13 (Ventura), not macOS 14 (Sonoma), so folks can fix this problem by updating their OS. Anyone who does not want to update macOS can use the keyboard workaround.

If anyone wants to explore further options to fix this without updating to macOS 14, I suggest trying to update the macOS SDK (Xcode) and watching https://gitlab.gnome.org/GNOME/gtk/-/issues/6697.

closed 2024-07-16 by mossmann