greatscottgadgets/packetry #130: Prepare Homebrew recipe (to be submitted to homebrew-core on release)

mossmann, 2024-07-14:


mossmann, 2024-07-16:
I have a basic, working formula. My plan is to release 0.1.0, then push the formula to an experimental tap (a repository containing the formula that makes it fairly convenient to use), invite folks to test it, and then push it upstream to homebrew-core after testing.

mossmann, 2024-07-17:
I realized that there is likely more benefit from going through the upstream CI and review process than there would be from testing in our own tap, so I am skipping the tap and working on getting this merged upstream: https://github.com/Homebrew/homebrew-core/pull/177539

martinling, 2024-07-25:
The Homebrew PR is now merged and it's possible to `brew install packetry`. :tada:

closed 2024-07-25 by martinling