greatscottgadgets/packetry #136: Add command-line option or separate capture tool

mossmann, 2024-07-17:
It should be possible to save a live capture to a file without the GUI.

martinling, 2024-07-17:
On Windows, this might have to be a separate binary, because as far as I'm aware it's not possible to have an executable that's a GUI application (i.e. can run _without_ opening a console window), but also outputs to the console if run from the command line.

This is confirmed by [Raymond Chen here](https://devblogs.microsoft.com/oldnewthing/20090101-00/?p=19643), although there are some ways to fake it listed  in [this stackoverflow thread](https://stackoverflow.com/questions/493536/can-one-executable-be-both-a-console-and-gui-application).

mossmann, 2024-07-17:
I think the typical solution is to have an additional executable like `packetry-cli` which I would be fine with.

martinling, 2024-07-17:
Yeah, but that reintroduces all the pain we had managed to remove in #105.

NullByteZero, 2024-10-31:
This will probably be a bit controversial question but why not create a CLI tool in python?

This would also allow us to import it as a library to our python projects and write our own decoders/analyzers which could process data in real time.

Its seems like a no-brainer to me since facedancer uses python and is extremely easy to use.

martinling, 2024-10-31:
The predecessor to Packetry was [ViewSB](https://github.com/greatscottgadgets/viewsb), which was written entirely in Python.

The key problem was that it was too slow. A saturated USB 2.0 high speed bus is pushing about 100,000 packets per second. The first stages of decoding really need to be in native code in order to keep up with the throughput.

What we might add at some point is a plugin API that lets higher level decoders be written in Python or some other scripting language, once the low-level packet capture is already processed and indexed.