greatscottgadgets/packetry #139: packetry --test-cynthion option (Windows) panics running Low-speed test

pointy56, 2024-07-22:
The --test-cynthion option of packetry for Windows completes the High-speed and Full-speed tests but panics during the Low-speed test with the following output to the console:

Testing at Low (1.5Mbps):

Opening analyzer device
Disabling test device
Enabling test device
Capture enabled, speed: Low (1.5Mbps)
thread '<unnamed>thread '' panicked at mainsrc\test_cynthion.rs' panicked at :src\test_cynthion.rs66::2763::
64called Result::unwrap() on an Err value: Failure in capture thread

Caused by:
    0: Sender was dropped
    1: oneshot canceled:

called Result::unwrap() on an Err value: Test device not foundnote: run with RUST_BACKTRACE=1 environment variable to display a backtrace

error: process didn't exit successfully: target\debug\packetry.exe --test-cynthion (exit code: 101)

(The version of packetry being used has been 'enhanced' to allow options when running from the Windows CLI)

Two USB Analyzer devices can be seen in Device Manager under 'Universal Serial Bus devices'; during the High-speed and Full-speed tests a USB Analyzer Test Device is also seen there.  During the Low-speed test a USB Analyzer Test Device is seen under 'Other devices' showing that no compatible driver is available:

[image]

pointy56, 2024-07-23:
Noticed in the Cynthion Discord forum that @mossman had seen what looks like a similar issue with the Apollo device.
I installed and ran Zadig, and it offered to install WinUSB (v6.1.7600.16385) for the problematic USB Analyzer Test Device.
I accepted this and following driver installation the Low-speed test now completes without issue.

martinling, 2024-07-23:
Thanks for the update.

The driver selection is supposed to happen automatically, and it uses the same WCID descriptors for low speed as it does at the other speeds, so I'm not sure yet what's going wrong there in the low speed case. It seems like it must be at least partially system-specific, as I didn't see this happen on my own Windows 10 test system.

We'll need to see if we can reproduce this, after clearing out cached descriptor info.

Good to know you found a workaround though.

martinling, 2024-07-31:
I've been able to reproduce this now.

martinling, 2024-08-08:
The panic is fixed in #164.

The cause of the test failure was not being able to open the test device.

This was due to the low speed version of the test device not being assigned the WinUSB driver.

That issue needs to be solved on the gateware side, and is now tracked here: https://github.com/greatscottgadgets/cynthion/issues/168.

closed 2024-08-09 by miek