greatscottgadgets/packetry #142: Re: Analyzer gateware is older (v0) than supported by this version of Packetry (v1)

capiman, 2024-07-28:
Duplicate to https://github.com/greatscottgadgets/packetry/issues/114,
but different request/solution:

I also received Cynthion and after some test on Linux, I switched to Windows.
I installed Packetry, which reported "Analyzer gateware is older (v0) than supported by this version of Packetry (v1)".
(I assume almost everyone using Packetry at least on Windows, will get this error).

Ok, I found out, that it seems, I must update FPGA. I need a bitfile analyzer.bit.

But how to build it? Found out, I need yosys, which I had not yet installed. After some tries I got path correctly
and it really built analyzer.bit.

Now I had the file, what to do with it?

```
martin@martin-ubuntu2024:~/.local/lib/python3.12/site-packages/cynthion$ cynthion flash analyzer
Traceback (most recent call last):
  File "/home/martin/.local/bin/cynthion", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/martin/.local/lib/python3.12/site-packages/cynthion/commands/cli.py", line 112, in main
    args.func(device, args)
  File "/home/martin/.local/lib/python3.12/site-packages/cynthion/commands/cynthion_flash.py", line 24, in cynthion_flash
    flash_bitstream(device, find_cynthion_bitstream(device, "analyzer.bit"))
  File "/home/martin/.local/lib/python3.12/site-packages/cynthion/commands/util.py", line 100, in flash_bitstream
    with open(filename, "rb") as f:
         ^^^^^^^^^^^^^^^^^^^^
TypeError: expected str, bytes or os.PathLike object, not NoneType
martin@martin-ubuntu2024:~/.local/lib/python3.12/site-packages/cynthion$
```

From this output it was not clear, where the file must be to be found.
I found code in util.py
( https://github.com/greatscottgadgets/cynthion/blob/07332f6bd62a463e087d8c636622aeca589d4e69/cynthion/python/src/commands/util.py#L60 )

I added a

```
print(bitstream_path)
```

to print, where it was searching for it:

/home/martin/.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev1D4/analyzer.bit

Perhaps I missed some steps, but even the CynthionPlatformRev1D4 was not existing.
I created it and put analyzer.bit there and then it worked:

```
martin@martin-ubuntu2024:~/.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev1D4$ cynthion flash analyzer                        /home/martin/.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev1D4/analyzer.bit
Updating FPGA configuration flash with 258949 bytes...
Operation complete!
martin@martin-ubuntu2024:~/.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev1D4$
```

Enhancement request: 
At least print where it is searching for the file by default.

Enhancement request: 
Could you add binary file analyzer.bit somewhere (e.g. website or one of the repos), so it is easier, at least for a beginner.
(so no need to download yosys, compiling by myself, just to do a first step with packetry)

I don't know if documentation would have existed and I have overseen it. If not, perhaps good to add at least a bit more in this area.
Or add more details, what to call exactly in which directory and what to get out or what to do in case of failure.

By doing all the above, I now can open Packetry on Windows and not getting the error message
"Analyzer gateware is older (v0) than supported by this version of Packetry (v1)" anymore.
(BTW: Perhaps you can add a URL into the error message (clickable?), to display what to do then.
I have done the complete update of Cynthion via Linux. Would all of this have worked even under Windows, like Yosys, building analyzer.bit, flashing this file...?)

capiman, 2024-07-28:
Perhaps Packetry could do the update completely by itself? Bringing all the needed file by itself?

martinling, 2024-07-28:
You don't need to build the analyzer bitstream from source.

See the [Getting started with Packetry](https://packetry.readthedocs.io/en/latest/quick_start.html#install-cynthion) section of the documentation.

All you should need to run is:
```
pip install cynthion
cynthion update
```
This will update the device with the latest pre-built analyzer bitstream.

capiman, 2024-07-29:
Thank you for your reply. It looks like I messed up my installation somehow.
I just uninstalled (python) cynthion and reinstalled it and now I have the following file on my disc (which were previously missing):

```
./.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev0D4/analyzer.bit
./.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev0D3/analyzer.bit
./.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev0D2/analyzer.bit
./.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev0D1/analyzer.bit
./.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev0D6/analyzer.bit
./.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev0D5/analyzer.bit
./.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev1D0/analyzer.bit
./.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev1D4/analyzer.bit
./.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev1D3/analyzer.bit
./.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev1D1/analyzer.bit
./.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev0D7/analyzer.bit
./.local/lib/python3.12/site-packages/cynthion/assets/CynthionPlatformRev1D2/analyzer.bit
```

Step by step I make progress...