greatscottgadgets/packetry #146: Windows binaries throwing potentially false positives for Win security features

mikepparks, 2024-07-29:
When I try to use the [zipped installer from the 0.1.0 release](https://github.com/greatscottgadgets/packetry/releases/download/v0.1.0/Packetry-Installer-for-Windows.zip), it returns a Microsoft Defender SmartScreen prompt (image 1).  As a precaution, I also tried to test the latest CI build from main's actions. On completed download, this binary triggers a Defender trojan detection (image 2). I could download the ``Binary for windows-latest.zip`` file, but it's a little concerning when both bundled installers are throwing potential virus warnings.

[image]
[image]

OS Name: Microsoft Windows 10 Pro
Version: 10.0.19045 Build 19045

mikepparks, 2024-07-30:
Doing some further testing, it seems that Action builds on ``main`` seem to start showing up this way as early as CI #503.

martinling, 2024-07-30:
The initial SmartScreen warning is to be expected because it's an unsigned binary.

The virus detection I haven't seen before, and I can't reproduce it here.

I've tried the latest CI #602, as well as the CI #503 build that you mentioned, and both install and run without alerts after selecting 'More Info' / 'Run Anyway' for the initial SmartScreen warning. I'm also on Windows 10.0.19045, with the virus & threat protection database updated just now.

So I see a couple of possibilities:

1. You have a trojan on your system, and it's infecting the file when you download it. We shouldn't rule this out.

2. The detection is a false positive, that has been fixed in the latest threat database, so I dont't see it. If so, then you should be able to avoid the false positive by updating (Settings -> Update & Security -> Windows Security -> Check for updates).

Searching for `Trojan:Script/Wacatac.B!ml` turns up a lot of GitHub issues and other discussions suggesting this may have been a common false positive as of a few months ago, but that's not really conclusive - it could be that users were seeing this on files they downloaded because their system was already infected and was modifying the files they'd downloaded.

If updating doesn't eliminate the warning, it would be interesting to see the `md5sum` of the `Windows installer.zip` files on your system.

mikepparks, 2024-07-30:
From [CI #595](https://github.com/greatscottgadgets/packetry/actions/runs/10143978602) ``Windows installer.zip`` gives me a sum of ``6e21d5fb550fbf79170f1f3e9923e9e5`` downloaded to a clean environment. My Defender DB is up to date as of today, and the PC has never had any sort of trojan in its lifetime.  It's interesting that it's doing this for ``main`` builds from CI #503 to CI #595. Grabbing files from CI #600 onward does not cause this, so perhaps it was a momentary bit of false positives.  Forgive the moment of Chicken Little.

martinling, 2024-07-30:
Interesting - CI #600 was triggered by the merge of PR #141. That PR did fix us doing things a little unusually for a GTK app, so maybe it saw that as suspicious, and now we're doing things in the more standard way it's less worried.

mikepparks, 2024-07-30:
Interestingly enough, it started to do this on newer CIs as well.

martinling, 2024-07-30:
I'm still not able to reproduce this here. Just tried the latest CI #620.

Windows reports that it's using:
```
Security intelligence version: 1.415.396.0
Version created on: 29/07/2024 19:08
```

mikepparks, 2024-07-30:
Yeah, I'm not sure what the issue is. I'm going to chalk it up to a false positive and do some digging on my machine.  Nothing else seems to be triggering this behavior on my end, and I don't want to make it seem like the program is the issue without solid evidence. Just wanted to provide info in case something upstream may have been an issue.

martinling, 2024-08-06:
I'm going to close this issue since I gather from Discord that you got things working, and without a way for us to reproduce the detection I don't think there's anything else we can do here.

If you or anyone else encounters this again, feel free to reopen.

mossmann, 2024-08-10:
I've just experienced "Couldn't download - Virus detected" while trying to download the Windows installer from https://github.com/greatscottgadgets/packetry/actions/runs/10322277671/artifacts/1795741895 (#166). Windows Defender thinks it is `Trojan:Script/Wacatac.B!ml`:
[image]

mossmann, 2024-08-10:
We are not the first to have this problem: https://answers.microsoft.com/en-us/windows/forum/all/overly-eager-heuristics-for-trojanwin32wacatacbml/6f2a72f3-3978-48ac-9fb7-fbe82c686ae3

Sounds like "machine learning" has probably learned to recognize one of our dependencies because it is also a dependency of some malware.

mossmann, 2024-08-10:
I'm not ready to recommend that anyone else do this, but I was able to work around the problem by temporarily excluding .zip and .msi files while downloading and running the installer.
[image]

mossmann, 2024-08-10:
```
Service Version: 4.18.24070.5
Engine Version: 1.1.24070.3
AntiSpyware Signature Version: 1.417.50.0
AntiVirus Signature Version: 1.417.50.0
```
I did `C:\Program Files\Windows Defender>MpCmdRun.exe -removedefinitions -dynamicSignatures` and `C:\Program Files\Windows Defender>MpCmdRun.exe -SignatureUpdate` a couple times and also tested with:
```
Service Version: 4.18.24070.5
Engine Version: 1.1.24070.3
AntiSpyware Signature Version: 1.417.52.0
AntiVirus Signature Version: 1.417.52.0
```
and got the same trojan detection.

martinling, 2024-08-11:
I think we have to consider whether this is our problem to solve.

This is an open source project for which every single line of code, as well as all the CI infrastructure for building it, is all available for everyone to inspect, reproduce and verify for themselves that it is free from malicious behaviour.

If someone trains a machine learning model to identify malicious code, and that model flags our binaries as malicious, then that is the absolute definition of a false positive, and the problem is with the model.

If the detection came with some details of which aspects of the binary were considered suspicious, then we would at least have the _option_ of altering our code to avoid triggering that particular false positive.

However, for obvious reasons, the report doesn't reveal anything about the detection method.

As such, there really is nothing we can do. The bug to be reported here is against the detection system.

closed 2025-02-20 by mossmann