greatscottgadgets/packetry #156: Add word wrap for longer data with optional ASCII decoding

Martyx00, 2024-08-02:
It would be nice to have the Packetry to wrap the data in the expanded view and also add option for automatic ASCII decoding. When longer data packets are captured, you cannot inspect them (unless you have really wide monitor).

[image]

martinling, 2024-08-02:
Our plan for this is to add a detail pane (#10) to display the full data for the currently selected item, in a separately-scrollable pane.

I don't think it's practical to show all data in the tree view without it becoming unnavigable. For valid transactions and packets there may be up to 1024 bytes of data, which would need about 20 lines of text at the width in your screenshot. And for transfers the length is unbounded. So although there may be cases where a bit of wrapping might suffice, I think the viable approach in the general case has to be to treat the tree view as an index, with the full data being shown elsewhere.

Note that there is already some automatic ASCII decoding. It kicks in [if at least half of the first 100 data bytes of an item are printable ASCII characters](https://github.com/greatscottgadgets/packetry/blob/7fba04aaee73712bc82f91a62b5d4783764dfbe8/src/capture.rs#L662-L684). This may not be the best heuristic, but it avoids us displaying an ugly mix of characters and escape sequences, in the case of binary data that is only coincidentally part-ASCII.

Martyx00, 2024-08-02:
Detail pane would be great, for ASCII decoding, I assume that a Wireshark Hexdump style data printing would be enough.

martinling, 2024-08-02:
Yeah, a hexdump-style view is what I've had in mind. There's quite a bit of work needed to implement that well though, such that it both scales nicely to the window width and loads data on-demand when scrolling through large transfers.

I may try to do something a bit more primitive in the near term.

closed 2024-08-09 by miek