greatscottgadgets/packetry #169: Saving a pcap takes minutes

claui, 2024-08-15:
On Packetry 0.2.0, saving captured packets to a pcap file usually takes minutes, while loading is instant.

## Steps to reproduce

1. Open Packetry.
2. Load any pcap file from your local file system. Observe that the file loads instantly, i.e. within tenths of seconds for a ~100MB pcap file.
3. Press the Save button on the toolbar. A file chooser comes up.
4. In the file chooser, pick a file name and hit Save.
5. Watch the progress bar near the bottom of the window:

[image]

## Expected behavior

Saving a ~100MB pcap file, originally recorded and saved using Packetry, takes **about three minutes** on a fast SSD.

## Actual behavior

Saving a ~100MB pcap file, originally recorded and saved using Packetry, takes **less than 30 seconds** on a fast SSD.

## Affected builds

This has occurred as long as I can remember. I only learned today via Discord that it’s not supposed to be that slow.

Builds with which I can reproduce the symptom:

- GSG’s own [v0.2.0](https://github.com/greatscottgadgets/packetry/releases/tag/v0.2.0) AppImage build

- the current [0.2.0-1](https://aur.archlinux.org/packages/packetry) package on the [AUR](https://aur.archlinux.org/), built from tagged upstream sources using `--release`

- the older 0.1.0-1 AUR package, built from tagged upstream sources using `--release`

- most (if not all) of the post-v0.1.0 tip-of-the-tree builds I’ve tried, built using the VCS package [packetry-git](https://aur.archlinux.org/packages/packetry-git) on the AUR via `--release`

## Notes

The above steps use a previously-saved pcap file for convenience. However, the same symptoms occur when saving a fresh Cynthion-recorded file.

Packetry says that the pcap file I’ve used has ~2.7M transactions and ~5.5M packets.

depili, 2024-08-15:
I can confirm this on macos with `cargo run --release` and 71827e2fcdcf227b3bbdab93b67cf4893f169e54

Saving a full speed 56 second capture of a ftdi-ft serial traffic with 4.13MiB reported size in packetry took around 5 minutes to save the 4,3M captured packets. This resulted in 75MB .pcap file.

martinling, 2024-08-15:
I've been unable to reproduce this here. I have a 300MB file I use regularly for benchmarking, with 1.5M transactions and 3.5M packets, and it saves in about 5 seconds on a release build. On a debug build it takes a couple of minutes.

martinling, 2024-08-15:
Turned out my file was a poor test because it used old dummy timestamp data. With a fresh capture I could reproduce the problem. Fixed in #170.

closed 2024-08-15 by martinling

claui, 2024-08-15:
Thank you for the fix @martinling!

PR #170 (2024-08-15): Use 24-bit rather than 16-bit increments for timestamps.
Using 16-bit deltas (2 bytes) in `CompactIndex` is a poor fit for nanosecond timestamp data, and causes the index to write a lot of blocks each time the delta range overflows, which also makes access inefficient. This is most noticeable when saving a capture, which requires reading all timestamps.

Using 24-bit deltas (3 bytes) works a lot better. I also tried 32-bit (4 bytes) but didn't see much difference in performance.

Fixes #169.