greatscottgadgets/packetry #172: Attempting to run packetry v0.2.1 AppImage on Ubuntu gives libpango undefined symbol error

x0rloser, 2024-08-20:
I am running Ubuntu v22.04 x64.

I downloaded this AppImage release.
[packetry-x86_64.AppImage](https://github.com/greatscottgadgets/packetry/releases/download/v0.2.1/packetry-x86_64.AppImage)

Attempting to run AppImage gives the following result:
```
❯ ./packetry-x86_64.AppImage
/tmp/.mount_packetiCLpnp/AppRun.wrapped: symbol lookup error: /tmp/.mount_packetiCLpnp/usr/bin/../lib/libpango-1.0.so.0: undefined symbol: hb_ot_layout_get_horizontal_baseline_tag_for_script
```

uname info
```
❯ uname -a
Linux xps 6.8.0-40-generic #40~22.04.3-Ubuntu SMP PREEMPT_DYNAMIC Tue Jul 30 17:30:19 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

antoinevg, 2024-08-20:
Can you please try running the "Linux AppImage binary" file at the bottom of this page:

https://github.com/greatscottgadgets/packetry/actions/runs/10469379702

x0rloser, 2024-08-20:
That now starts running successfully, and finds my cynthion device.
However when I click on the "record" button I get an msgbox popup error "No such file or directory (os error 2)". There is nothing extra printed to the terminal console that I ran it from.

I updated my cynthion device to the latest firmware today via "pip install cynthion" and then "cynthion setup" and then "cynthion update".

If I close packetry and then run it again, I can click on "record" and don't get any error message this time. However I also don't get any packets captured. It shows "0 devices, 0 endpoints, 0 transactions, 0 packets" in the status bar at the bottom. It shows this while recording and after stopping the recording.

I have the target-C port plugged into a windows pc and the target-A port plugged into a usb device. I have captured packets for this usb device before in windows using wireshark, so I know that there is usually usb traffic. I also heard the "usb device plugged in" sound from the windows PC and I can see the usb device present in device manager on the windows pc. So the passthru of the usb packets between the target-C and target-A ports is working, but the packet capturing is not.

antoinevg, 2024-08-20:
Can you please tell me a little more about the USB device you are trying to capture?

By default packetry starts in USB High-speed (480Mbps) capture mode and, unless the device you are trying to capture is also USB High-speed you will not see any captured packets.

You can set the Speed in packetry by stopping capture and selecting the "Speed" dropdown in the toolbar.

x0rloser, 2024-08-20:
I was testing with a HID control pad.

It seems the incorrect speed selection is indeed the problem with 0 packets captured.
I changed it from 480Mbps to 12Mbps and I started to see packets being captured.
As a test I also tried 1.5Mbps and it gave a bunch of "invalid groups" packets and then a thread panicked with the following error. Do you want me to create a separate issue for this error?

```
❯ ./packetry-x86_64.AppImage
Capture enabled, speed: High (480Mbps)
Requesting capture stop
Capture disabled
Capture enabled, speed: Low (1.5Mbps)
thread '<unnamed>' panicked at src/usb.rs:320:25:
index out of bounds: the len is 0 but the index is 0
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

I am not sure how to get the environment variable set in a way that displays the full backtrace.
i tried the following, but neither worked:
```
❯ RUST_BACKTRACE=1
❯ ./packetry-x86_64.AppImage
```
```
❯ RUST_BACKTARCE=1 ./packetry-x86_64.AppImage
```

The docs don't mention anywhere about needing to select the right speed for it to work. I know it seems obvious once you point it out, but it isnt obvious to someone new to using packetry. I assumed that using the highest speed would also catch any lower speed. So it is probably worth pointing out the need to select the correct speed in the "getting started docs".

I do wonder if cynthion could detect the speed of the current device and "auto-select" the correct speed, or maybe give some warning msg about the selected speed not being the expected speed for the device.,

antoinevg, 2024-08-20:
> It seems the incorrect speed selection is indeed the problem with 0 packets captured. I changed it from 480Mbps to 12Mbps and I started to see packets being captured. 

🎉

> As a test I also tried 1.5Mbps and it gave a bunch of "invalid groups" packets and then a thread panicked with the following error. Do you want me to create a separate issue for this error?

Yes please, that would be greatly appreciated!
 
> The docs don't mention anywhere about needing to select the right speed for it to work. I know it seems obvious once you point it out, but it isnt obvious to someone new to using packetry. I assumed [...]

Great suggestion, thank you! 

Could you please add this to the issue you created in the Cynthion repo: https://github.com/greatscottgadgets/cynthion/issues/177

> I do wonder if cynthion could detect the speed of the current device and "auto-select" the correct speed, or maybe give some warning msg about the selected speed not being the expected speed for the device.,

Under development: https://github.com/greatscottgadgets/cynthion/issues/53

x0rloser, 2024-08-21:
Upon looking back it turns out the speed selection is indeed mentioned in the packetry docs on page docs/source/quick_start.rst, seems I just missed it.

Adding a link to the packetry docs at the end of the cynthion page might be useful to get to this info however, so I have added that suggestion to #177

PR #173 (2024-08-27): Add libharfbuzz to Linux AppImage
During packetry startup on Ubuntu `22.04.x` (and possibly others) the system will attempt to load `libharfbuzz` even though we are not linking to it directly, with the result that the system's version gets loaded instead:

```
openat(AT_FDCWD, "/tmp/.mount_packetjchMhG/usr/bin/../lib/libharfbuzz.so.0", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libharfbuzz.so.0", O_RDONLY|O_CLOEXEC) = 4
read(4, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
```

Because the version of `li [... cut]
closed 2024-08-27 by miek