greatscottgadgets/packetry #174: Choosing Low-Speed with Full-Speed device connected causes unhandled exception

x0rloser, 2024-08-21:
test setup: Using cynthion with packetry on ubuntu 22.04 and with a full-speed (12Mbps) usb hid device connected to cynthion. Using most recent cynthion fw and packetry sw versions.

If I select the speed to be Low-Speed and click on record, after about 3 seconds I get the following error on the terminal console that I launched packetry from:

```
❯ ./packetry-x86_64.AppImage
Capture enabled, speed: Low (1.5Mbps)
thread '<unnamed>' panicked at src/usb.rs:320:25:
index out of bounds: the len is 0 but the index is 0
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Sometimes it will also display "Invalid groups" in the traffic window multiple times.

martinling, 2024-08-23:
Please try this with `RUST_BACKTRACE=1`, the backtrace would be useful.

If the wrong speed is selected then the analyzer can end up reading garbage from the PHY. That garbage is essentially fuzzing the decoder here, and has triggered a code path that leads to an out of bounds index operation.

Some combinations of true/selected speed will not result in garbage being captured (e.g. if LS/FS capture is selected on an HS bus, the HS signalling levels are below the logic high threshold).

Once we have auto speed detection in the analyzer this should be less of an issue.

x0rloser, 2024-08-27:
I tried setting ```RUST_BACKTRACE=1``` but it didn't seem to work.
I am using the AppImage prepared specially to fix #172, so maybe it is not a normal build.
This is the AppImage I am using: https://github.com/greatscottgadgets/packetry/actions/runs/10469379702

This is my attempt at enabling backtrace.
(I am running on Ubuntu 22.04)

```
❯ RUST_BACKTRACE=1
❯ echo $RUST_BACKTRACE
1
❯ ./packetry-x86_64.AppImage
Capture enabled, speed: Low (1.5Mbps)
thread '<unnamed>' panicked at src/usb.rs:326:52:
index out of bounds: the len is 2 but the index is 2
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

miek, 2024-08-28:
> I tried setting `RUST_BACKTRACE=1` but it didn't seem to work.

Just `RUST_BACKTRACE=1` alone only sets the variable for the current shell. To also set it for any other processes started from that shell, you need to run `export RUST_BACKTRACE=1`

martinling, 2024-08-28:
This may well be fixed by one of the commits on #182.

closed 2024-08-29 by miek