greatscottgadgets/packetry #175: Clicking record will overwrite existing data without warning the user

x0rloser, 2024-08-21:
When you have captured a packet log in packetry, it will overwrite that log without wanring if you then click on the record button.

It may be beneficial to add a warning msgbox to let the user know that they will lose their current packet log if they proceed. This is similar to how wireshark currently treats this situation.

The msgbox could have a tickbox to "do not warn me anymore this session".
Also if the packet log was already saved to disc somewhere, then perhaps the warning msg would not be needed.

x0rloser, 2024-08-21:
If adding a warning msg, perhaps also add one for when the user attempts to close the app with an unsaved packet log.

martinling, 2024-08-22:
I think the best solution here is probably to add support for having multiple captures open at once (e.g. with a tabbed UI), and then the record button can always start a new capture, without having to dispose of the previous one first.

I'd specifically like to avoid a dialog preventing the record button taking immediate effect.

When debugging hard-to-reproduce issues, it can sometimes take hours or even days to reproduce a problem. Hitting record on a capture device is like pressing the shutter on a camera - it needs to work immediately, or the moment you were trying to capture is lost.

When exiting with unsaved capture data, a dialog makes sense.

x0rloser, 2024-08-22:
I hear you on the hard to reproduce a problem thing! :)
Which is why it is important to not accidentally lose/overwrite a capture.

mossmann, 2024-08-27:
In the meantime we can at least document the current behavior: #183