greatscottgadgets/packetry #178: Packetry crashes when resizing

x0rloser, 2024-08-22:
Like the title says I can crash packetry by just resizing the main window.
This happens without opening any log or having any cynthion hw connected to my machine.
There is nothing printed to console if I start it from the console, and no msgboxes or anything are displayed.
The program just quits out immediately.

Done using packetry v0.2.1 release from git for Windows.
I am using Windows 11 x64 with all latest windows updates.
I do have multiple monitors, but it crashes even when I keep the bounds of the window within my main (middle) monitor.

Replication steps:
1) start packetry (either from windows start menu or from command line)
2) position mouse cursor on the bottom right corner until you see the "resize" mouse cursor show up.
3) click and hold the resize cursor the move the mouse anticlockwise circles like you are drawing a clockface, starting at the 12 at the top and moving thru the 9, 6, 3, and back to 12 again.
4) The program crashes and disappears for me everytime after 1 or 2 anticlockwise circles.

x0rloser, 2024-08-22:
I tried this on the Linux build running on Ubuntu v22.04 and I cannot get it to crash.

martinling, 2024-08-22:
If you run `packetry-cli` from the console, you should get some debug output, and also a backtrace if you run `set RUST_BACKTRACE=1` beforehand. My guess is that it will be some internal crash in GTK's Windows backend.

x0rloser, 2024-08-22:
I did try with packetry-cli from console already and got no debug output.
I also tried just now after setting that env variable and also got no output.
Here are the commands i tried just in case I am doing something wrong:

```
C:\Program Files\Packetry\bin>set RUST_BACKTRACE=1
C:\Program Files\Packetry\bin>echo %RUST_BACKTRACE%
1
C:\Program Files\Packetry\bin>packetry-cli
C:\Program Files\Packetry\bin>
```

It returned to the comand prompt when the program crashed.

mossmann, 2024-09-05:
@x0rloser Was this from the binary installer or compiled from source? Have you tried release 0.2.2?

I've been unable to reproduce the problem with 0.2.1 or 0.2.2 from the binary installer.

martinling, 2024-09-08:
I'm also unable to reproduce this with the 0.2.2 installer.

x0rloser, 2024-09-10:
I just tried with the v0.2.2 installer for windows from github releases page, and it still crashes in the same way.

Previously I had used the v0.2.1 installer for windows from the github releases page.

martinling, 2024-09-10:
I haven't been able to try this on Windows with multiple monitors yet.

Are you able to reproduce this if you disconnect all but one monitor?

martinling, 2024-09-10:
Also: are you able to reproduce this with any of the GTK example programs? I suspect that this is going to turn out to be a GTK bug that's specific to the Windows backend on multi-monitor systems.

x0rloser, 2024-09-11:
I disabled all my monitors except one (via display settings -> disconnect this monitor), and set the remaining monitor to the standard "landscape" mode. I tested again and I get the same crash when resizing.

Are the GTK exmaple programs available anywhere prebuilt for windows? I don't have the time just now get a working gtk dev env sorted.

martinling, 2024-09-11:
The [gvsbuild](https://github.com/wingtk/gvsbuild) project has [binary releases](https://github.com/wingtk/gvsbuild/releases) of the full GTK stack for Windows, but I've just checked and they don't include the example programs unfortunately.

x0rloser, 2024-09-13:
thanks for the link - i've reported the issue there, and while it does seem somewhat nvidia driver related, im still hoping it will get fixed so i don't lose whatever i am doing in packetry due to reszing the window xD

x0rloser, 2024-09-13:
I reported this issue to the gnome/gtk gitlab project.
It seems the problem is due to "NahimicOSD" dll, and is a known problem.

For more info see: https://gitlab.gnome.org/GNOME/gtk/-/issues/7003

The fix that worked for me was:

Open up services.msc as Admin, and scroll down to 'Nahimic service' ... right click 'Stop', and then right click 'Properties'.  While in the 'General' tab, make sure 'Startup type' says 'Disabled'.  Then in the 'Recovery' tab, make sure the 'recovery failures' are all set to 'Take No Action' . And then click OK when you are all set.  Click on the pictures below to see the whole image. Good luck !

After doing this it packetry no longer crashes for me during resizing.

For the record, I didn't have to reboot or anything after doing this.

closed 2024-09-13 by martinling