greatscottgadgets/packetry #179: A failed `libusb1::bulkRead()` operation can cause Packetry to display a `PID Malformed does not indicate a direction` error.

antoinevg, 2024-08-22:
When I am capturing a transaction and `libusb1` is performing a `bulkRead()` operation that fails with a `LIBUSB_ERROR_PIPE` exception that causes the host program to exit abruptly the following occurs:

1. Packetry pops up the following dialog:  `PID Malformed does not indicate a direction`
2. Subsequent attempts to start a new capture pops up the following dialog: `Could not be opened for exclusive access`
3. Restarting Packetry allows me to resume capturing.

When I simply swallow the `LIBUSB_ERROR_PIPE` exception in the host program Packetry does not display any dialogs and continues to capture without any problems.

martinling, 2024-08-23:
Running with `RUST_BACKTRACE=1` should get you a backtrace in the dialog.

Could you also try using the command-line capture mode from #136 to get a capture of the traffic that triggers this. That mode just dumps to pcap without running the decoder, so should avoid triggering the bug.

antoinevg, 2024-08-29:
Tx!

Exceptions:

[image]
[image]

Traffic capture using `cynthion-cli`: [cynthion.pcap.gz](https://github.com/user-attachments/files/16796472/cynthion.pcap.gz)

The capture file handily reproduces the `PID Malformed` exception when loaded into packetry which says all kinds of cool things about packetry's design :-)

Finally, a little bit more info:

This was happening when the host issued a `CLEAR HALT` on the bulk endpoint which was not handled by the device.

martinling, 2024-08-29:
Thanks, I'll take a look.

martinling, 2024-08-29:
The first error is fixed in #182. The packet that caused the problem was an SOF packet with a bad CRC. It was preceded by two IN packets with bad CRCs, but those didn't trigger the same bug:

[image]

I expect the second error is just a side effect of the first error: the device cannot be opened because there is a handle to it still left open by the capture/decoding process which failed.

closed 2024-08-29 by miek