greatscottgadgets/packetry #186: ISO transfers all under single polling group.

AndrewCapon, 2024-09-11:
So I can see the order of my bulk transfers, the polling and then the "Unidentified transfer".

All my ISO packets though are contained in single "Polling" groups so I get something like : "Polling 89064 times for unidentified transfer on endpoint 36.3 OUT" and "Polling 89064 times for unidentified transfer on endpoint 36.3 IN".

If I expand the groups I see the transactions.

Why are these not displayed with the other transfers?
[image]

martinling, 2024-09-11:
This happens because the enumeration process wasn't captured, so Packetry hasn't seen the configuration descriptor request, which would include the endpoint descriptors in the response.

Without seeing the endpoint descriptors, Packetry doesn't know what the type of endpoint 2 is, so it describes transfers made on it as `unidentified`.

If you start the capture before you connect the device, you should get the descriptor tree in the devices pane, and these transfers should then be correctly identified.

AndrewCapon, 2024-09-11:
Hi,

Thanks for the reply.

The device is being started after the capture has started though.

AndrewCapon, 2024-09-11:
I also noticed not all the configurations are shown in the right pane, is this normal?

martinling, 2024-09-11:
The single configuration being shown seems correct to me. The device descriptor has `bNumConfigurations` set to 1, indicating that there's only one configuration, and the host only requests that one descriptor.

The thing with isochronous transactions being shown in polling groups is a bug, though. I think the state machine is wrong here and it's expecting an ACK, which it shouldn't do in the isochronous case. I'll see about a fix.

AndrewCapon, 2024-09-11:
Sorry used wrong word, I meant Alternate. So interface 1 alternate 0, interface 1 alternate 1, interface 1 alternate 2.

martinling, 2024-09-11:
Yeah we may be doing something wrong there too. I'll take a look.

martinling, 2024-09-12:
The issue with alternate interface settings is fixed in #189, which now includes a clip of your first capture as a test case.

[image]

AndrewCapon, 2024-09-12:
> The issue with alternate interface settings is fixed in #189, which now includes a clip of your first capture as a test case. [...]

That's great, thanks for that :)

AndrewCapon, 2024-09-12:
> I have some work in progress on fixing the isochronous transfers too, but that's going to take a bit longer to get right.
> 
> One of the more tricky issues involved is that in the case where we _haven't_ seen the endpoint descriptors, a transaction with just IN or OUT followed by DATA0 or DATA1 is ambiguous. It _could_ be a successful isochronous transaction, but it could also be an incomplete bulk/interrupt transaction that wasn't ACKed or NAKed by the host when it should have been.

Yeah. I see your point.

martinling, 2024-09-19:
The issues with handling isochronous transactions are fixed in #192, which I'd appreciate your review on. I think that your two captures shared above are now interpreted correctly.

Note though that the result is not all that different - the isochronous transactions on each endpoint are still placed in one group, but that group is now labelled as an isochronous transfer rather than as a polling group.

AndrewCapon, 2024-09-19:
Hi Martin,

Thanks very much for that.

So the iso transactions are not interspersed with say the bulk transactions? 

Thanks

Andy

AndrewCapon, 2024-09-19:
Hi Martin, 

Thanks for the info, I wondered why all the SOFs were grouped, the same reason?

Looks like I am probably after the transaction or packet view.

I have gone back to using my old analyser, I will build and try your new version when I get a chance...

Thanks

Andy

closed 2024-09-21 by miek

martinling, 2024-09-21:
> Thanks for the info, I wondered why all the SOFs were grouped, the same reason?

Yes, SOFs are gathered at two levels:
- Consecutive SOF packets on the wire, without other traffic in between are placed in a group equivalent to a transaction.
- All SOF groups in the capture are themselves placed in a group equivalent to a transfer.

The benefit of this becomes more obvious in the transaction view, or the interleaved view:
[image]

AndrewCapon, 2024-10-27:
Sorry, I have been meaning to post here to say thanks for all the updates you have done. Packetry is so much more useful for me now. Nice job :)