greatscottgadgets/packetry #187: OSX Problems - Cannot set speed and about box not working.

AndrewCapon, 2024-09-12:
Hi Guys,

I have built packetry for osx, I cannot set the speed though.

When I click to pulldown the menu I see this in the log:
```
(packetry:1146): Gtk-WARNING **: 08:12:06.108: Broken accounting of active state for widget 0x7fd86c6145e0(GtkLabel)

(packetry:1146): Gtk-WARNING **: 08:12:06.108: Broken accounting of active state for widget 0x7fd86c614470(GtkBox)

(packetry:1146): Gtk-WARNING **: 08:12:06.108: Broken accounting of active state for widget 0x7fd86c158d60(GtkListItemWidget)

(packetry:1146): Gtk-WARNING **: 08:12:06.108: Broken accounting of active state for widget 0x7fd86c1589d0(GtkStack)

(packetry:1146): Gtk-WARNING **: 08:12:06.108: Broken accounting of active state for widget 0x7fd86c1587e0(GtkBox)

(packetry:1146): Gtk-WARNING **: 08:12:06.108: Broken accounting of active state for widget 0x7fd86c158660(GtkToggleButton)

(packetry:1146): Gtk-WARNING **: 08:12:06.108: Broken accounting of active state for widget 0x7fd86c158410(GtkDropDown)

(packetry:1146): Gtk-WARNING **: 08:12:06.108: Broken accounting of active state for widget 0x7fd86c15b300(GtkBox)

(packetry:1146): Gtk-WARNING **: 08:12:06.108: Broken accounting of active state for widget 0x7fd86c20c0f0(GtkBox)

(packetry:1146): Gtk-WARNING **: 08:12:06.108: Broken accounting of active state for widget 0x7fd86c20c6f0(GtkCenterBox)

(packetry:1146): Gtk-WARNING **: 08:12:06.108: Broken accounting of active state for widget 0x7fd86c20bd00(GtkRevealer)

(packetry:1146): Gtk-WARNING **: 08:12:06.108: Broken accounting of active state for widget 0x7fd86c20b860(GtkActionBar)

(packetry:1146): Gtk-WARNING **: 08:12:06.108: Broken accounting of active state for widget 0x7fd86c6137f0(GtkBox)

(packetry:1146): Gtk-WARNING **: 08:12:06.108: Broken accounting of active state for widget 0x7fd86c209fd0(GtkApplicationWindow)

```

I can click on the new setting in the pulldown but the value does not hold, it stays at 480.

Also the about box is not working.

Anyone any ideas?

Thanks

Andy

AndrewCapon, 2024-09-12:
Ah, I can set it using the keyboard, so pull down the menu, arrow keys then return.

martinling, 2024-09-12:
This has come up before - see #129. It's a GTK/macOS issue rather than a Packetry-specific bug.

It's being [tracked here](https://gitlab.gnome.org/GNOME/gtk/-/issues/6697) in the GTK issue tracker. So far it seems that it goes away if you update to a newer version of macOS.

AndrewCapon, 2024-09-12:
Ah, thanks Martin.

closed 2024-09-12 by AndrewCapon