greatscottgadgets/packetry #211: Can't connect to cynthion

sehicfai, 2024-11-13:
I have following problems with my cynthion device and packetry. I hope you can help me with that or maybe show me where I can get
the help for this. Problem Packetry shows me following message: "This device is not usable because: Failed to claim interface". So I checked the documentation provided on the website and check the hardware connectivity:

C:\Users\X>cynthion info
Cynthion version: 0.1.7
Apollo version: 1.1.0
Python version: 3.12.2 (tags/v3.12.2:6abddd9, Feb  6 2024, 21:26:36) [MSC v.1937 64 bit (AMD64)]

Found Apollo stub interface!
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\X\AppData\Local\Programs\Python\Python312\Scripts\cynthion.exe\__main__.py", line 7, in <module>
  File "C:\Users\X\AppData\Local\Programs\Python\Python312\Lib\site-packages\cynthion\commands\cli.py", line 92, in main
    args.func(args)
  File "C:\Users\X\AppData\Local\Programs\Python\Python312\Lib\site-packages\cynthion\commands\cynthion_info.py", line 19, in cynthion_info
    if ApolloDebugger.print_info(force_offline=args.force_offline, out=logging.info,
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\X\AppData\Local\Programs\Python\Python312\Lib\site-packages\apollo_fpga\__init__.py", line 436, in print_info
    out(f"\tBitstream: {device.product} ({device.manufacturer})")
                        ^^^^^^^^^^^^^^
  File "C:\Users\X\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb\core.py", line 875, in product
    self._product = util.get_string(self, self.iProduct)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\X\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb\util.py", line 313, in get_string
    raise ValueError("The device has no langid"
ValueError: The device has no langid (permission issue, no string descriptors supported or device error)

When using the pc of my colleague it work fine. I am not sure where my mistake is. I tried following things:

- pip install --upgrade cynthion: Works fine (Requirements already satisfied)

- C:\Users\X>cynthion setup --check : I get:
usage: cynthion [-h] command ...
cynthion: error: argument command: invalid choice: 'setup' (choose from 'run', 'flash', 'build', 'update', 'info')

- Different USB Connector: No differences

miek, 2024-11-13:
Another user had a similar issue here: https://github.com/greatscottgadgets/cynthion/issues/169#issuecomment-2282295209 and there's some steps posted in that thread which may fix it. Could you try those out and let us know if it helps?

sehicfai, 2024-11-18:
I tried the steps mentioned there but I get an error by "cynthion flash analyzer".   

File "C:\Users\X\AppData\Local\Programs\Python\Python312\Lib\site-packages\usb\backend\libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 2] Entity not found

I get the same error when I try "cynthion update --bitstream" but not with cynthion update --mcu-firmware. There I get "Updating device firmware with 13876 bytes...
Operation complete!".

Since I pressed the program button I cant find the Cynthion as USB Analyzer" or "WinUSB Device", but only as "Cynthion Apollo Debuger".

sehicfai, 2024-11-18:
Okay it's working now. This steps weren't needed:

- Press PROGRAM
- cynthion info for confirmation
- cynthion flash analyzer

What worked was just:

- Open Device Manager and find pair of USB devices called "USB Analyzer" or "WinUSB Device"
- Right-click and "Uninstall" both devices
- Unplug Cynthion
- Plug in Cynthion
- cynthion info for confirmation

I uninstalled only the both "USB Analyzer" nothing else.

Thank you!

closed 2024-11-18 by sehicfai