greatscottgadgets/packetry #213: Can't save capture to file

vulpes2, 2024-12-02:
Perhaps there's an off by one bug somewhere, I can't save my capture to a file at all with v0.4.0 (appimage).
[image]

martinling, 2024-12-02:
That sure does look like an off-by-one bug.

I haven't seen this before and would love to have a way to reproduce it, but obviously that's difficult if you can't save the capture.

Is it reproducible if you capture the same device again? If so could you try with `RUST_BACKTRACE=1` set in the environment? That should at least get a backtrace which might indicate where the bug is.

vulpes2, 2024-12-02:
Did a new capture with the same device, and I can't reproduce it anymore. Will keep an eye on it and post the backtrace if I ever come across it again.

closed 2025-09-17 by martinling