greatscottgadgets/packetry #214: Class level decoding

jolo1581, 2024-12-05:
Hello,

is it possible to have an option to decode the USB stream to Class level like Total Phase Beagle 480?
It is nice to see the datastream, but most time I need to look on class level. So I can directly see what is going on.

martinling, 2024-12-05:
In the long run, decoding class-specific requests and data structures is possible and I'd consider it in scope for the project.

There are some small things already done, e.g. if you look at the request for a HID report descriptor, the detail pane will show the structure of the report.

But to do this sort of work across all classes is a huge job, because of how many classes there are and how many different data formats are used in each of them. I've not personally used Total Phase's products, but I'm sure that they must have put a great deal of work into that feature of their software.

I think that the best solution is probably not for us to write Rust code for all classes directly, but to use some kind of definition language in which the data formats used by each class can be expressed more succinctly, and which Packetry can load, interpret and use at runtime. That would make it a lot faster to translate each class specification into a form that Packetry can work with, and would enable more people to contribute to that effort.

If we can find a sufficiently intuitive way to do this, then users would also be able to use that same approach to describe the data formats used by vendor-specific devices, which would be a great feature to have too.

ardnew, 2025-02-15:
The [sigrok](https://sigrok.org/wiki/Main_Page) project uses a similar [DSL](https://en.wikipedia.org/wiki/Domain-specific_language)-based approach. It enables a broad user base to define high-level serial protocols for a tool that otherwise simply records I/O voltage levels.

They've separated that effort into a library [libsigrokdecode](https://sigrok.org/wiki/Libsigrokdecode) that defines the so-called [protocol decoder API](https://sigrok.org/wiki/Protocol_decoder_API) for consumption by front-end tools. Their [HOWTO](https://sigrok.org/wiki/Protocol_decoder_HOWTO) gives more detail.

They've adopted Python3 for this purpose — probably for popular appeal and adoption — which personally makes my toes curl a bit.

More on point, though, is some of the USB protocol decoders they've already defined:

- [usb_signalling](https://sigrok.org/gitweb/?p=libsigrokdecode.git;a=blob;f=decoders/usb_signalling/pd.py;h=484e5e58441fcb2d504b74aec4a5f29c87fbacf3;hb=HEAD)
- [usb_request](https://sigrok.org/gitweb/?p=libsigrokdecode.git;a=blob;f=decoders/usb_request/pd.py;h=2f5abccd28befaafe4acde6bfc6340453d0981af;hb=HEAD)
- [usb_packet](https://sigrok.org/gitweb/?p=libsigrokdecode.git;a=blob;f=decoders/usb_packet/pd.py;h=769db9a3a5ab92868cb4107f948ec8a16f51e1cc;hb=HEAD)
- and even [usb_power_delivery](https://sigrok.org/gitweb/?p=libsigrokdecode.git;a=blob;f=decoders/usb_power_delivery/pd.py;h=3788924a546d05013e10e8d89b3f364e95904630;hb=HEAD)

martinling, 2025-02-24:
The sigrok example is one I'm very familiar with actually - I was one of the core developers on that project for a few years. I wasn't around when the libsigrokdecode API was initially being defined, but I got to see a lot of the regrets later.

The big problems as I saw them were:

1. It required doing everything in Python, and that was slow. Low-level decoders initially had to process every single sample one at a time in Python. This worked well enough for decoding short captures of things like I2C and SPI at low sample rates, which was most of the initial use cases, but it couldn't scale. We were able to improve things a little with the v3 decoder API, which allowed low-level decoders to say "skip samples until X happens" and have that part be done in native code, but that only helped so much.

2. Although we had a nice system for protocol layering (e.g. `usb_request` stacks on top of `usb_packet`, which in turn stacks on the `usb_signalling` decoder), the means of passing data between layers was Python objects. That made it impractical to port the lowest level decoders (e.g. SPI) into native code to make them faster, without breaking compatibility.

3. Decoders always just started reading at the start of the capture and progressed linearly through it. There wasn't any way to zoom into an area of interest and just decode that part, with the decoder being able to "look back" at earlier events as necesary.

4. By the time these mistakes were fully recognised, there were already 100+ decoders in the repository, and it was too late to make major changes without rewriting them all.

Because of that experience, I'm very concerned with making sure that when we start adding plugin decoders to Packetry, we're doing so in a form that's going to avoid similar issues.

Ideally, I would like that to not be procedural code at all, but rather just some kind of definition language which sets out the various requests and data formats used by a given class or device. That leaves the runtime implementation free to evolve as needed.

If that's not viable, and we do need procedural code in plugin decoders, then the API for that needs to work in the same way that Packetry does - with individual items being decoded on-demand for performance, rather than batch processing the whole capture.

When plugin decoders need information from earlier in the capture (e.g. to decode a data transfer, we'll probably need to know information from back when the device was enumerated and set up), that should either happen by querying the capture database as needed, or perhaps through decoders being able to register events of interest for Packetry to tag them during its high-speed decoding pass.

ardnew, 2025-02-26:
Another approach that would avoid this sort of batch processing is to define the plugin interface as a lexer that emits standard USB packet-oriented tokens. 

The user-defined decoder, then, would be a plain-text grammar composed of these tokens, which defines the desired capture sequence (I'm imagining something akin to eBNF). Packetry would be responsible for lexing the captures, generating the parser per given grammar(s), feeding tokens from lexer to parser, and finally presenting the capture according to the grammar.

This approach scales naturally, as grammars could be defined in terms of other grammars.

I'm no rust programmer, but a cursory search yields a few parser libraries — [pest](https://pest.rs/) and [nom](https://github.com/rust-bakery/nom) — that look well-suited to this task.

I like the idea, but maybe its a bit too generalized? Just another idea.

martinling, 2025-02-26:
I don't really see how that approach is applicable here.

There is a part of USB protocol decoding that does look very much like parsing against a grammar, but that part is common to the low-level protocol and is already complete - you can see it in [`Decoder::transaction_status`](https://github.com/greatscottgadgets/packetry/blob/55a37191788abaf491d27dd49ff4fc90a5b6a796/src/decoder.rs#L96-L239), which parses transactions from of sequences of packets, and then at the next level up in [`Decoder::group_status`](https://github.com/greatscottgadgets/packetry/blob/55a37191788abaf491d27dd49ff4fc90a5b6a796/src/decoder.rs#L280-L484), which parses transfers and other groupings from sequences of transactions.

Different USB devices and classes don't use different sequences of packets. What varies is basically just payload data.

The jobs that class and device specific decoders are going to need to do are:
1. Take a class-specific descriptor for which we have a type ID and raw bytes, and provide a brief description and/or a breakdown of the contents.
2. Take a control request (i.e. [`struct ControlRequest`](https://github.com/greatscottgadgets/packetry/blob/55a37191788abaf491d27dd49ff4fc90a5b6a796/src/usb.rs#L1170-L1176)) and return a human-readable summary of its meaning, and/or a breakdown of whatever fields are contained in its payload data.
3. Take data transfers made on an endpoint and give a human-readable description of the meaning of the payload (e.g. for USB mass storage: this data is blocks 1234 to 4567 from the storage device, or for USB audio: this is 16-bit PCM data for channel 1).

martinling, 2025-02-28:
When I said that different devices don't use different sequences of packets, I was responding to your description of using "USB packet-oriented tokens".

Of course grammars and parsing can be applied to descriptors and so forth, but that's a process that starts at the point when we've already extracted the descriptors from the capture. Packets are no longer relevant. It's not useful to try to define things in terms of nested grammars from the packet level up.

Similarly, we don't need a grammar to distinguish a `Configuration` descriptor from an `Interface` descriptor, because those are standard descriptors which we already parse.

There's a clear delineation throughout the USB protocol between three categories of things:

1. Standard: things that are defined in the USB standard itself. These are implemented in the Packetry core, and we don't need to reinvent them.

2. Class: things that are defined in class specifications. These we'd like to be able to farm out to plugin decoders that would each claim some specific class by its ID.

3. Vendor: Things that are defined by individual vendors. These we'll want to farm out to plugin decoders that would each claim one or more specific devices (primarily by VID and PID).

Because of the way USB is structured, we will always know exactly which of these categories things are in, even if we don't know anything about that specific class or device. For instance, all descriptor formats that are class-defined have type IDs in the range `0x20..0x3F`. So we'll know if it's a class-specific descriptor, and we know from the class and subclass fields in the device/interface/function descriptors exactly which class ID is responsible for understanding it, even if we've never heard of that class. The same is true for control requests: there are specific ranges for standard, class and vendor requests.

The whole protocol is deliberately structured around making this sort of division of responsibility easy, so that operating systems can know what drivers to assign, etc. It makes sense for us to follow these mechanisms because that corresponds to how the protocol is used in practice.

So we will always know _which_ decoder we need to hand things off to for decoding, and we'll either have a matching decoder available or we won't. We don't need to invent new matching mechanisms.

The big questions here are just about what form we write the individual decoders in. I have some ideas which I'll follow up with.

martinling, 2025-02-28:
The architecture that I am thinking about at the moment is to use [Kaitai Struct](https://kaitai.io/) as a DSL for binary data formats, and [Lua](https://www.lua.org/about.html) as an embedded scripting language for higher level interpretation.

- Kaitai Struct is the most serious attempt I have found at a user-friendly, language-independent mechanism for describing binary data structures, and it has been demonstrated to be suitable for a wide range of [existing binary formats](https://formats.kaitai.io/).

- Lua is probably the best-suited of several options for an embedded scripting language. It is specifically designed to be small and easily embeddable, and is widely used for that purpose, including in Wireshark.

Packetry would make use of a `plugins` directory containing `.ksy` files defining Kaitai schemas for data structures,  `.lua` files that implement class and device specific decoders, and some additional plain-text configuration files that specify how all these should be matched to class IDs, VID/PID values, request numbers and so forth.

At runtime, Packetry would use the Kaitai schemas to parse descriptors, requests, etc and display the decoded fields in the detail pane and the device hierarchy. This approach would also enable graphical views as discussed in #217. Importantly, no procedural plugin code would be required for this part to work.

For generating human-readable summaries of control requests and data transfers, Packetry would call into plugin decoders written in Lua that run in an embedded interpreter. These Lua decoders would have access to any already-parsed data structures associated with the transfer and the device, and would also be able to make queries to the capture database via some internal API, to obtain information about earlier requests. The decoder would be able to stash its internal state for a device, so that when called again to decode some other transfer on the same device, it would be able to make use of information already gathered.

The Lua decoders would primarily be responsible for generating human-readable text summaries (e.g. "Setting RF frequency to 1.23 GHz"), but the approach could also be extended to other functionality, e.g. tagging payload data types like images, video or audio so that they could be displayed in the GUI.

There would be no guarantees about what transfers a decoder would see, or what order it would see them in - it would have to be able to start from anywhere and decode things in whatever order it is asked to.

Packetry would monitor the `plugins` directory, e.g. by inotify, and would automatically re-run parsing and decoding for visible items upon relevant changes to the schemas, decoders or matching configuration. On reloading a decoder, all existing state for it would be discarded.

antoinevg, 2025-02-28:
> The architecture that I am thinking about at the moment is to use [Kaitai Struct](https://kaitai.io/) as a DSL for binary data formats, and [Lua](https://www.lua.org/about.html) as an embedded scripting language for higher level interpretation.

+1 on Kaitai and Lua, they're both great tools for this job. 

Rust support for Kaitai is pretty sparse but it's not a hard DSL to implement from scratch if we need to. Rust/Lua on the other hand is a pretty mature space and we might even want to consider using the https://github.com/kyren/piccolo interpreter.

martinling, 2025-02-28:
I think implementing it from scratch is exactly what we'd need to do. We're not going to be compiling Rust parsers for Kaitai, we're going to need to be able to:
- read in a schema at runtime
- parse bytes against that schema
- iterate over the fields of the resulting structure
  - to display them in the UI
  - to generate an equivalent Lua object
All of that is well outside existing use cases. If we were willing to have a runtime dependency on the Kaitai struct compiler then we could use that to generate parsers in Lua and run them in the interpreter, but that might be a pain from a distribution point of view.

ardnew, 2025-02-28:
> - read in a schema at runtime
> - parse bytes against that schema
> - iterate over the fields of the resulting structure
>   - to display them in the UI
>   - to generate an equivalent Lua object

This is exactly the approach I was suggesting(!) in my first comment:

> The user-defined decoder, then, would be a plain-text grammar composed of these 
> tokens […]. Packetry would be responsible for lexing the captures, generating the
> parser per given grammar(s), feeding tokens from lexer to parser, and finally 
> presenting the capture according to the grammar.

Though, I certainly didn't know which layer would be best to emit tokens (lexemes) from. 

From what I understand from your comments, emitting bytes would be redundant given that a certain amount of structure is always guaranteed (e.g., `Configuration` and `Interface` descriptors).

Thus, totally agree that using descriptors then as your tokens would be the lowest-level any decode grammar would ever need.

And to be clear, I don't think Packetry would actually need to `parse bytes against that schema` itself. Kaitai generates a parser that will do that for you. Packetry wouldn't even need to define the data structures in Rust for storing the parsed schema, since they too are constructed by Kaitai according to the YAML-based grammar input.

But, I think we're absolutely on the same page. Love the work so far, and can't wait to see if this gets pursued!

martinling, 2025-02-28:
I don't really see how descriptors would be being treated as tokens here. The whole concept of tokens as I understand it is that you have _sequences_ of tokens, and parsers that recognise languages from those sequences using a grammar.

But there's no such sequence to descriptors. Descriptors can be fetched from a device in any order and their meaning is the same regardless of that order. We're not parsing anything from a grammar of descriptor tokens.

What we need to parse is the raw bytes of binary structures that form class/device specific descriptors, request payloads, data payloads, etc.

> And to be clear, I don't think Packetry would actually need to parse bytes against that schema itself. Kaitai generates a parser that will do that for you. Packetry wouldn't even need to define the [...]

The point is that we want to do this at runtime. I don't want users to have to recompile Packetry to incorporate new schemas or make changes to them. It should be possible to iterate on this while Packetry is running.

So Kaitai generating Rust structures from schemas doesn't help us. We need to parse into to some kind of flexible data structure that's constructed at runtime.

That could mean generating a Lua parser from the schema and running it in the interpreter, and using Lua objects as the runtime representation, but as it stands that would require us to call the Kaitai struct compiler at runtime to produce the Lua code. I'm not so keen on that, since it complicates the packaging and distribution situation.

So this may be something we write our own code for, even if we adopt the Kaitai DSL.

ardnew, 2025-02-28:
> But there's no such sequence to descriptors. Descriptors can be fetched from a device in any order and their meaning is the same regardless of that order. We're not parsing anything from a grammar of descriptor tokens.

Imagine a token being any single character in the regular expression `.*`. This describes a grammar in which the sequence, or ordering, of tokens does not matter. The meaning of each token is the same, regardless of the order they were consumed.

Now, if you generalize the notion of a token from a single character to a single USB descriptor, maybe you can see how grammars can recognize many different sequences of descriptors `fetched from a device in any order and their meaning is the same regardless of that order`.

martinling, 2025-02-28:
OK, I see your point. I think that might be a good model for thinking about how we match decoders to data structures: as regular expressions that define the range of structures they know how to interpret.