greatscottgadgets/packetry #226: Voltage Capture

Heisenberg-192, 2025-03-05:
Can we export the voltage data while the USB is communicating with the host ?

martinling, 2025-03-05:
This is on the roadmap but there's quite a bit to do.

There's some work that needs to be done on the Cynthion analyzer gateware first, to make it collect the voltage/current data and multiplex it into the data stream going to the host.

Then my plan is to add an oscilloscope-style graphical view to Packetry, where the voltage and current waveforms can be plotted along with the decoded USB traffic so that you can see how the two relate.

Heisenberg-192, 2025-03-05:
Yehh, thats exactly what i want for my objective. Do you think there are any other alternatives that i can consider to build it ? My goal is to extract the respective analog data of the usb packets.

martinling, 2025-03-05:
There's various approaches that you could use with additional hardware. But any approach that uses the Cynthion alone is going to require doing some development work on the analyzer gateware. If you're up for pursuing that it would be a very welcome contribution.