greatscottgadgets/packetry #233: Packetry not showing data while fuzzing

sankethj, 2025-05-15:
My setup:
Cynthion in analyser mode.

My aim was to fuzz the target using umap. So I am using goodfet board for fuzzing. Cynthion is in-between target and goodfet board to log the actual usb packets using analyser. I can see packets also being logged in packetry, but it only logs for some time like 200 seconds. Later no fuzzing packets are being logged in packetry.
When I try to stop packetry, it says error broken pipe.

Is there any limitations for logging in cynthion or packetry?

I can include any logs if needed.

martinling, 2025-05-15:
There shouldn't be any limitations on capturing with Cynthion and Packetry in that scenario.

The only real limitation on capture is the HyperRAM buffer space on the Cynthion side. With a saturated High Speed (480Mbps) bus, that can fill up quicker than Packetry can read out the data (which happens at the same speed, plus overheads).

But since you're working with a GoodFET board you're presumably only generating Full Speed (12Mbps) traffic.

Perhaps your fuzzing has turned up a bug on our side?

sankethj, 2025-05-15:
That might be the case. The fuzzing might have triggered something on cynthion end, as there was no crash or anything from the actual target side. I will try to debug this weekend and post the details here.