greatscottgadgets/packetry #246: Feature request: aarch64 build

XenoKovah, 2025-07-30:
I'm using Ubuntu 24.04 aarch64 on an ARM-based Mac, and it looks like the Packetry app image is x86-64 only.

This is a request to provide an aarch64 build as well on the Releases page. (Because other Cynthion tooling is working up until this point.)

martinling, 2025-07-31:
So yeah the good news is that our main build & test workflow works out of the box on the new `ubuntu-24.04-arm` runner. 

The bad news is that the AppImage build process has `x86_64` hardcoded in a whole bunch of places and it's going to take a while to figure out all the right incantations to change that.

antoinevg, 2025-08-01:
The appimage build happens inside a `x86_64` debian buster docker container that's executed by the `ubuntu` runner. 

To do ARM builds there are basically two options:

1. Set up a ARM cross-compile toolchain on the existing debian image. This could be challenging given that a) gtk can be a chore to compile at the best of times and b) there's quite a long list of dependencies.
2. Adapt the existing `x86_64` workflow for an `ubuntu-24.04-arm` runner and use the Debian Buster ARM image for the Dockerfile. Main changes would probably be:
    - Use ARM versions of the various appimage tools
    - Update references to library paths
    - Potentially deal with some gtk-4 compilation breakage

It's pretty straightforward but there may be more than a few unexpected yaks and fiddlies on the way.